NubeIO/rubix-point-server #187: generic point HTTP PATCH

NubeDev, 2021-01-21:
generic point has no support for HTTP PATCH.

Needs to be added

Shiny380, 2021-01-22:
### Allow `write_value` to write to `generic_point`
Generic point needs to be reworked as it's general scope/function is changing
for now just implement following changes:
#### TODO:
- Change `generic_point`.`writable` to `True`
- if `write_value` is changed in `PATCH` request -> update `point_store` with `write_value`

Shiny380, 2021-01-22:
#### Note: Check following issue
https://github.com/NubeIO/rubix-point-server/issues/189
If this issue isn't much more work it would be better to close this one and work on that instead

RaiBnod, 2021-01-22:
Ya, sure we can close this down and work on the other ticket. The other ticket more descriptive.

PR #193 (2021-01-28): PATCH generic value support
Resolves: #189
Resolves: #187

### Summary:

Made an `PATCH` API available for POSTing value for Generic writable points. URL endpoint example:

```bash
curl -X PATCH http://localhost:1515/api/generic/points/<uuid>/value -H "Content-Type: application/json" -d '{"value": <value>, ...}'
```
closed 2021-01-28 by NubeDev