NubeIO/rubix-point-server #205: Let user set point type

NubeDev, 2021-02-01:
Let user set point type

- string
- float
- int

RaiBnod, 2021-02-02:
It should be only for generic point.

RaiBnod, 2021-02-02:
@NubeDev while we use the point data type as `string`, we need to change on followings:

On `point_store` and `point_store_history` tables:

- `value` from `float` to `string` (we store `1` as `"1"`, `"value"` as `"value"`, `"20.12"` as `"20.12"`)
- `value_original` `float` to `string`

& also on `InfluxDB` we will store value as `string` previously it was storing as `float`.

Moreover, our `priority_array` array fields values are also `float`, they needs to convert it on `string`.

Please confirm on it before proceeding the changes.

Shiny380, 2021-02-02:
Related #80 
This is a major change and should be left until later after better discussion
More important issues to be worked on atm

closed 2021-02-02 by Shiny380