NubeIO/rubix-point-server #293: Please check if these work as expected

NubeDev, 2021-04-10:
[image]

CanuckMarc, 2021-04-11:
No, scaling doesn't work as expected.  See screenshot markup below.  It seems like `Scale Min` and `Scale Max` are reversed.

Scaling function should be based on remap function:
``` 
if (value > inMax) return outMax;
else if (value < inMin) return outMin;
return ((value - inMin) / (inMax - inMin)) * (outMax - outMin) + outMin;
```

[image]

NubeDev, 2021-04-11:
See code 
https://github.com/NubeIO/lora-raw/blob/e13c28810111f0a1833fb214228d5456df0ff001/src/models/model_point.py#L99

CanuckMarc, 2021-04-11:
`apply_scale` function is correct, so it must be the implementation that is incorrect.  My best guess is that `Scale Min` and `Scale Max` are reversed in the function call.  

Should check the code that sets the values of `self.scale_min` and `self.scale_max` are pulling from the correct platform columns  
HERE: https://github.com/NubeIO/lora-raw/blob/0399090cfa506aab3c0b96b24e6d2ffaa1f78fde/src/models/model_point.py#L84

Also, as per https://github.com/NubeIO/lora-raw/issues/80 Scaling shouldn't be needed for Droplets (only for microedges)

CanuckMarc, 2021-04-12:
1. This is trying to add scaling in the `LORA service`.  Seems to not like the math function or something. I have deleted the `data` file and re-added my droplets. 
https://www.awesomescreenshot.com/video/3353291?key=6ac611605cd08c5e1fcdd79881b482ba

2. In the `Rubix Points` the scaling function looks to be working well now.  My only minor suggestion would be to switch the `scale max` and `scale min`, from a user perspective, it should always be `min` then `max` when entering.  
[image]

3. As per my markups here: https://github.com/NubeIO/rubix-point-server/issues/295.  I think the scaling should have a `Scaling Enable` checkbox, and only show the `input`/`scaling` factor fields when the checkbox is Enabled.

CanuckMarc, 2021-04-21:
Update:
1. Scaling is not working on LORA points.  See video below.
2. GUI settings for the scaling values should be in this order `Input Min` `Input Max` `Scale Min` then `Scale Max` currently they are reversed where they are `Max` then `Min`. 

https://www.awesomescreenshot.com/video/3469866?key=1b041b57462e9f17ad55f0fb6648f3c7

RaiBnod, 2021-04-23:
Now, 
- scaling is working well
- max, min positions are reversed

Hence, closing down the ticket.

closed 2021-04-23 by RaiBnod