NubeIO/rubix-point-server #296: LORA to RUBIX point mapping issues

CanuckMarc, 2021-04-11:
1. Point multi-select needs work.  The functions of the `select all` and `select all on device` interfere with each other.  In my opinion, `Select All` and `Clear All` should just be buttons (not checkboxes).  

2. `Writeable` should be enabled by default:
[image]

3. Values should be sent to the Rubix Points when created:
[image]

4. `Auto add Devices, Points, and Mappings` option doesn't work:
[image]

CanuckMarc, 2021-04-21:
Update:
Entering the point mapping works well, but the result in the `rubix service` has issues.

See video of the process below.  The created `rubix` points do not have `writeable` ticked, and when I try to enable it manually, it gives me errors.  Once these points are writeable, we need to confirm that the value that comes through is the correct `value` with the `math expression` applied, and not the `raw value`. 

https://www.awesomescreenshot.com/video/3469811?key=0e1361489df4a64f656179bac02a279b

RaiBnod, 2021-04-29:
`writable` is by default `true` now and other existing issues are also fixed.

Hence, closing down this ticket.

closed 2021-04-29 by RaiBnod