NubeIO/rubix-point-server #324: Add INTERVAL&COV option for histories

CanuckMarc, 2021-04-29:
We need to add `COV & INTERVAL` as an additional option to the rubix point `History Type`.  This configuration would save a history based on both the `COV Threshold`, and based on the `History Interval`.  

For example:
Point has `COV & INTERVAL` option selected on `History Type`.  If the value changes by the `COV threshold`, then a history is logged.  A history log is also saved at the `History Interval`, on the same point.

CanuckMarc, 2021-05-17:
@RaiBnod The `COV & INTERVAL` option is not available in the latest builds (plat and point-server both @1.7.0)

[image]

NubeDev, 2021-05-17:
This is a UI issue @CanuckMarc

CanuckMarc, 2021-06-03:
@RaiBnod  The `INTERVAL AND COV` option isn't working.  I have a bunch of Boolean points that are set for `COV AND INTERVAL` histories, with `History Interval` set to 60 and `COV Threshold` set to 0.  It is only getting histories logged on the `INTERVAL` (60 mins), but not when there is a COV (true-false transition).

PR #348 (2021-05-17): Add COV_AND_INTERVAL history type support
Resolves: #324

### Summary

- Add COV_AND_INTERVAL history type support
PR #367 (2021-06-03): Release/1.7.3
Resolves: #324

### Includes

- Add priority to MQTT message
- Issue fix: value were not writing on COV (#324)
closed 2021-06-03 by RaiBnod