NubeIO/rubix-point-server #337: Have priority array support fallback values

CanuckMarc, 2021-05-05:
It would be good if we could assign a fallback value to each rubix point.  This would mean if the priority array 1-16 is all `null` then the point would have its fallback value.  The fallback value can't be written from anywhere else except in the rubix point service in the platform.

RaiBnod, 2021-06-06:
@CanuckMarc what should be the fallback value number?

CanuckMarc, 2021-06-06:
Good question. For now, let's just call it priority 16 if it uses the fallback value (or an input at 16). That is how it works in Niagara. But we may need to change it to another number (like 17) later.

RaiBnod, 2021-06-06:
It's good to have 16 priority array only, as the convention. Ok then, we are changing it to, 16 priority with a value of 16 at fallback.

CanuckMarc, 2021-06-06:
Ya exactly. Keep only 16 priority levels. And it is at level 16 when using the fallback value, OR when it has an input at level priority 16.

RaiBnod, 2021-06-06:
**Just confirming:**

- We are going to have a fallback value at priority array 16.
- And for the fallback value, we will have an extra field as `fallback_value` and it should be of value `16` and we will pick that value on nullability.

CanuckMarc, 2021-07-06:
@RaiBnod I think you guys might have misunderstood the intent of the `fallback` value.  Or I don't understand how it is supposed to work. 

It seems like the rubix points are falling back to a VALUE of `16` when all 16 priority slots are `null`.  I do not see a `fallback value` property that can be set.  This is not correct.  

There should be a UI property for each rubix point called `fallback value` and if all the priority array slots are `null` then the `value` for the rubix point should be whatever the `fallback value` is.  By default the `fallback value` should be `null`.

Let me know if you have questions.

RaiBnod, 2021-07-12:
@CanuckMarc this is yet to be implemented on point server on frontend side: https://github.com/NubeIO/rubix-plat/issues/687#issuecomment-863877101

PR #371 (2021-06-11): Issue/cov point
Resolves: #369 
Resolves: #337 
### Summary:
  -  Added `fallback_value` (default=16) for points.It will be the value of priority_array `_16` when all 16 prior_array are null.
closed 2021-12-21 by RaiBnod