NubeIO/rubix-point-server #352: Add point mapping by name

NubeDev, 2021-05-17:
Let user do point mapping by name

CanuckMarc, 2021-05-30:
[image]

RaiBnod, 2021-05-31:
After we've been playing with mapping much, we've a good idea on existing mapping improvements.

**Scope:**

- Add `mapping_state` for showing mapping link. Status are: `MAPPED`, `BROKEN`.
- Refactor mapping schema as this:
    ```python
    class LPGBPointMapping(ModelBase):
        """
        lora_point <> generic_point | bacnet_point
        """
        __tablename__ = 'mappings_lp_gbp'
    
        uuid = db.Column(db.String, primary_key=True)
        point_uuid = db.Column(db.String, db.ForeignKey('points.uuid'), nullable=False)
        mapped_point_uuid = db.Column(db.String(80), nullable=True, unique=True)
        point_name = db.Column(db.String(80), nullable=False)
        mapped_point_name = db.Column(db.String(80), nullable=True)
        type = db.Column(db.Enum(MapType), nullable=True) # GENERIC, BACNET
        mapping_state = db.Column(db.Enum(MappingState), nullable=True) # MAPPED, BROKEN
   ```
- Refactor POST request, It should be via uuid and name:
  - uuid: it will be called when we are doing normal mapping, we take just two uuids with `type` and fills the names dynamically
  - name: it will be called when we are doing import/export, we take just two name with `type` --name will be colon (:) delimited
- Add an API to update the `mapping_state`:
  - Change names from `uuid` if names are update
  - Update `mapping_state` as per it's current state; `BROKEN` when the mapped point is no more in the table
- Migration: delete older table and re-create table again.

PR #365 (2021-07-16): Add point mapping by name
Resolves: #352 
Resolves: #377 
### Summary 
- Migration: modified `mappings_mp_gbp` table and transfer data to modified table.
- Modified endpoints
  -  PATCH `api/mappings/mp_gbp/uuid/<uuid>`
    ```
       -d
       {
          "point_uuid":<point_uuid>,
          "mapped_point_uuid":<mapped_point_uuid>,
          "type":"<type>"
       }
    ```
    
- Added new endpoints.
  - POST `api/mapping/mp_gbp/uuid` 
    ```
       -d
       {
          "point_uuid":<point_uuid>,
          "mapped_point_uuid":<mapped_point_uuid>,
          "type":"<type>"
       }
    ```
 [... cut]
closed 2021-07-16 by RaiBnod