NubeIO/rubix-point-server #364: multiple uses of `Schedule Name`

CanuckMarc, 2021-05-27:
We need to find another name for the Schedules that are added in the Rubix Platform.  `Schedule Name` is already used as a property of a schedule entry.  We could maybe use `Schedule ID` or `Schedule Container Name`. 

These should not be called `Schedule Name`:
[image]

[image]

These are the `Schedule Names`:
[image]

CanuckMarc, 2021-05-27:
@RaiBnod 

ADDITIONALLY: In the schedule topic (`<client_id>/<site_id>/<device_id>/rubix/points/listen/schedules/<name|uuid>/<schedule_name|schedule_uuid>`) in the current use cases, this topic can be shortened, it doesn't need the last part.  Maybe in future it is needed.

For now, this topic is what is needed:
`<client_id>/<site_id>/<device_id>/rubix/points/listen/schedules/<scheduleID|uuid>`  Where `scheduleID` is the `Schedule ID` or `Schedule Container Name` as discussed above. 

This topic should return the full Schedule JSON as it has been for quite a while.  It can contain multiple `Schedule Names`.

CanuckMarc, 2021-06-06:
@RaiBnod Could you please let me know how to use these schedule topics?  I've had the MQTT console open with a topic of `+/+/+/rubix/points/listen/schedules/+/+` but I can't see any values coming through.  I've tried creating a new schedule in the Platform, and posting to a schedule from Grafana. But I don't see any messages on that topic.  Please assist.

RaiBnod, 2021-06-07:
**Schedule list** (it returns the list of available schedule topics):
  - `<client_id>/<client_name>/<site_id>/<site_name>/<device_id>/<device_name>/rubix/points/value/schedules`

**Schedule Value** (after adding the schedule details on the Grafana, we will get the value of that Grafana schedule on this topic):
  - `<client_id>/<site_id>/<device_id>/rubix/points/listen/schedules/<name|uuid>/<schedule_name|schedule_uuid>`
  
If you still got the issue, please ping me and we'll do debug.

closed 2021-06-07 by CanuckMarc