NubeIO/rubix-point-server #366: Make default point mapping for temperature have default COV Threshold of 0.01

CanuckMarc, 2021-06-03:
Make default point mapping for temperature have default COV Threshold of 0.01.  Currently it is 0.1

RaiBnod, 2021-06-06:
@CanuckMarc where is that actually. I have created a droplet on LoRa and mapped it with Rubix Generic and it's showing `COV_THRESHOLD` as `0.01`, could you please confirm?

1. LoRa Points
[image]

2. LoRa > Rubix Points Mapping
[image]

3. Created Rubix Points from Mapping
[image]

Possibly you may have updated `COV_THRESHOLD` to `0.1` on the LoRa side before creating the mapping. Or, you have updated `COV_THRESHOLD` to `0.1` on `Rubix Points` after it's been created.

NubeDev, 2021-06-06:
This is not a valid issue.
This could be done in wires plat maybe only if tags are used

closed 2021-06-06 by NubeDev