NubeIO/rubix-point-server #372: Rubix Schedules do not push out on local MQTT

CanuckMarc, 2021-06-08:
@RaiBnod 
It seems like the Rubix Schedules are only available on the cloud MQTT broker.  I can't see any schedule values being pushed through on the `localhost` MQTT broker on the RC. 

For schedules on my local RC I can see the schedule updates on `+/+/+/rubix/points/listen/schedules/+/+` subscribed to the `nubeio` cloud broker (password `UDL9qQqxA6KZWaf9`).  But when I'm subscribed to the same topic on the RC `localhost` broker, I don't get anything.

RaiBnod, 2021-06-09:
That's how schedules are working.

- Our Rubix Point Server is responsible for creating the schedule list. And the list of schedules are pushed on the topic:
   - `<client_id>/<client_name>/<site_id>/<site_name>/<device_id>/<device_name>/rubix/points/value/schedules`
- Grafana connects to the cloud MQTT, and it sees the list of schedules from that subscription
- We create a schedule
- We select a topic from the list from the schedule
- Schedule values get published on cloud MQTT (we are using cloud MQTT as storage for this schedule's structure and data)
  - `<client_id>/<site_id>/<device_id>/rubix/points/listen/schedules/<name|uuid>/<schedule_name|schedule_uuid>`

In our current flow, cloud MQTT has no idea of the local MQTT connection. So, it can't communicate with local MQTT.

**We will make it available for local MQTT as well.**

CanuckMarc, 2021-06-09:
@RaiBnod 

My issue is that I need to get the schedule JSON (`value`) into the wires nodes.  I don't know how to subscribe to the cloud MQTT broker from a node.  The cloud MQTT broker details aren't in the `.env` file where the other `host`/`port` details come from. 

It would be ideal if the schedule JSON updates (from cloud MQTT broker) can be saved in the `point server` (so it persists on reboot).  Also, a matching topic should be created in the local MQTT broker so that it can be subscribed to from the wires nodes.

CanuckMarc, 2021-06-09:
FYI: We will eventually need to build a GUI for setting schedules (like in grafana) in the platform.  So the schedule JSON will need to be stored and modified by the point server, or by a seperate schedule service, whichever makes more sense.

CanuckMarc, 2021-06-17:
@RaiBnod can I get a status update on this?  Need schedules to local MQTT broker.

RaiBnod, 2021-06-17:
Now, we are making that server schedule available on the local broker as well.

@enjuthulung look at this:
Listen to could broker on schedule topics and publish the value on the local broker.

CanuckMarc, 2021-06-22:
@RaiBnod  @enjuthulung 
I have just been testing the schedule messages that are coming in over the local MQTT.  But, i'm getting a JSON parse error on the message:
```
topic: "c-10nube10/s-b6ba52ce/d-0dca2082/rubix/points/listen/schedules/name/TESTING_SCHEDULE"
payload: "{'weekly': {'131783ec-d3f8-4472-b3a3-67515bdd778a': {'name': 'HVAC', 'days': ['sunday', 'monday', 'tuesday', 'wednesday'], 'start': '22:00', 'end': '04:00', 'value': 0, 'color': ''}}}"
```

I think the problem is that the schedule message uses single quotation marks `'` instead of double quotation marks `"`.  

Could you please have a look?

CanuckMarc, 2021-06-22:
@RaiBnod  
Now there seems to be a `b` at the begginning of the JSON, and a couple of extra `'` (beginning and end) of each schedule message.
```
topic: "c-10nube10/s-b6ba52ce/d-0dca2082/rubix/points/listen/schedules/name/TESTING_SCHEDULE"
payload: "b'{"weekly":{"131783ec-d3f8-4472-b3a3-67515bdd778a":{"name":"HVAC","days":["sunday","monday","tuesday","wednesday"],"start":"22:00","end":"23:00","value":0,"color":""}}}'"
```

CanuckMarc, 2021-06-22:
@RaiBnod see my previous comment. Issue is still present.

RaiBnod, 2021-06-22:
@CanuckMarc I've updated that same version, check it. It should be fine.

PR #376 (2021-06-18): Make server schedule available on the local broker.
Resolves: #372 
### Summary:
  -  Added `cloud` field on MQTT setting (by default False).
  -  Listen to cloud broker(cloud=true) on schedule topics and publish the value on the local broker(cloud=false).
PR #382 (2021-06-22): Release/1.7.6
Resolves: #372

### Summary

- Fix: schedule dispatch JSON (need to sync JSON)
PR #383 (2021-06-22): Release/1.7.6
Resolves: #372

### Summary

- Fix: schedule dispatch JSON (need to sync JSON)
closed 2021-12-21 by RaiBnod