NubeIO/rubix-point-server #398: [MODBUS] `Write Value Once` points still need to `read` values.

CanuckMarc, 2021-07-19:
When a modbus point `Write Value Once` property is set, the point should write it's value (immediately on COV), then the point should do only `READ` requests (which update the point value, but not the `write` value).  

Currently the points with `Write Value Once` property set, will write the point value (good!), but after that, the point value remains the write value, whereas it should update to the current value of the point (by doing a `read` action, along with the rest of the `read` points).

Shiny380, 2021-07-28:
this is different to the `write value once` property
this is a different issue we need to tackle regarding "bi-directional writing"
`write value once` property needs to function as is and a new property "bi-directional write" (or something like that) needs to be implemented. this new values needs to override `write value once`

another maybe cleaner option is replace these two props with a single enum prop `write type` with the following modes:
- write values always
- write value once / write on COV
- bi-directional writing

Bi direction writing needs to be discussed in depth further

Shiny380, 2021-11-01:
refer to this issue for updates description:  
https://github.com/NubeIO/flow-framework/issues/284