NubeIO/rubix-point-server #419: Implement Schedules on Backend and Frontend

NubeDev, 2021-08-18:
This issue may be in the wrong repo, so feel free to move it.

This issue is for discussion on how to implement schedules within the rubix platform UI and the associated backend support.

General Requirements:

User can view and set schedules (as is currently done in the grafana widget) from the platform UI. Schedules need to be able to be set from grafana also.
User can link configured schedules to modbus (IO module) applications. This would happen via a python/go application that sets the required modbus registers for the schedule functions on the IO modules.
User can link schedule outputs to rubix points. This would require doing the schedule checking function in the backend instead of in wires. This is intended to allow for mass changes to which schedule is operating a system without editing wires for each system.

CanuckMarc, 2021-08-18:
Rubix Schedule -> Modbus (for IO Modules):

- WEEKLY Rubix Schedule (configured in the Dashboard or Platform UI) is broken up into the daily periods, and the data is written to the many modbus registers that configure the onboard schedules of the IO Modules. 

- The IO module programs can have multiple schedules (different modbus register sets), so we need to be able to select which Rubix Schedule we want to be linked to which IO Module schedule.

CanuckMarc, 2021-08-18:
Rubix Schedule Linking:
- We will need a UI and backend support for linking Rubix Schedules to both Rubix Points, and to IO Module Schedules.  

- Each Rubix Schedule should be able to be linked to many Rubix Points, and many IO Module Schedules at once.